prusa3d/PrusaSlicer #8928: Seam in concentric top/bottom infill when arachne is used

ochm, 2022-09-21:
### Description of the bug

When the concentric top/bottom infill is used and the Arachne generator is used, the begin and the end of extrusion on top/bottom infill is not set on diagonal path as was set in classic. When the classic generator is used, the start and end of extrusion is OK.
Arachne generator result:
[image]

Classic generator result:
[image]

### Project file & How to reproduce

The project: [test.zip](https://github.com/prusa3d/PrusaSlicer/files/9618997/test.zip)

To reproduce:
Change the Generator to Arachne, slice and check top and bottom tool path (seam in infill)
Change the Generator to Classic, slice and check top and bottom tool path (seam in infill), compare with arachne result.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Windows 10

### Printer model

Prusa MK3 S

kubispe1, 2022-09-23:
Somehow it realy starts to plan loop from the middle. I try for simple cube and there is a same result:
[image]

ochm, 2022-09-25:
The seam position in infill seems to be randomly selected, sometimes after slicing is position OK, but for example after move of object and reslicing is the seam on other position.

rsdrakh, 2023-04-05:
Something to consider for concentric:
In case there is another layer coming (e.g. with details), hide the seam under that.

[image]

war4peace, 2023-09-20:
I experience the same issue, came here to report it but found this one. Still happens with PrusaSlicer 2.6.1

closed 2026-09-29 by prusaslicer-bot[bot]