prusa3d/PrusaSlicer #8930: If top interface layers is set to 0 - interface layers still generated

T3P3, 2022-09-21:
### Description of the bug

If the "top Interface layers" setting in the support material settings is set to 0 then no interface layers should be generated, however they are (in PS 2.5.0, other versions not checked).

This is noticeable if you have a different extruder selected as the interface extruder, in this example tool 4is in grey and tool 3 is in orange.

[image]

the support material interface extruder is set to 3

[image]

but no support material interfaces should be generated due to this setting

[image]

I noticed this because I got an unwanted tool change from tool 4 to tool 3. It may have been an issue in previous versions but if you are using the same tool for everything, including the support material interface extruder, then the issue is masked.

### Project file & How to reproduce

### Project file

[example_ps_interface_extruder_issue_project.zip](https://github.com/prusa3d/PrusaSlicer/files/9620460/example_ps_interface_extruder_issue_project.zip)

### To reproduce
Setup a printer with more than one tool

Set the support material interface extruder to a different extruder from the others

Set the interface layers to 0

Examine the gcode preview and/or the generated gcode for the unwanted tool change.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-win64

### Operating system

Windows 10

### Printer model

Modified e3d Toolchanger

rtyr, 2022-09-22:
Reproduced. Thanks.

It seems some part of sparse supports is misrepresented as interfaces.

RR-Rico, 2023-04-29:
Does still apply to the newest 2.6 alpha.

closed 2026-09-29 by prusaslicer-bot[bot]