prusa3d/PrusaSlicer #8931: Split to objects crashes Prusa Slicer

aclucca, 2022-09-22:
### Description of the bug

Split to objects crashes Prusa Slicer 2.5

### Project file & How to reproduce

open elsa.3mf
click split to objects
-> crash!

[slicer25](https://user-images.githubusercontent.com/97547887/191797211-b17e6f10-4062-4668-8df6-8ce1058a845e.PNG)
[elsa.zip](https://github.com/prusa3d/PrusaSlicer/files/9627566/elsa.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Windows 10 64bit

### Printer model

Prusa MK3

YuSanka, 2022-09-23:
Thanks for your report.
Bug is caused by zero volume of the one of meshes of a splitable object.
Fix will be a part of next release.

FidelCapo, 2022-10-03:
This issue should be fixed and the fix will be part of the next release, probably 2.6.0.
Closing issue.

closed 2022-10-03 by FidelCapo