prusa3d/PrusaSlicer #8932: Hang while slicing interlocking chainmail model

Lupo-Grigio, 2022-09-23:
### Description of the bug

When trying to slice this model, on an intel macbook pro 2019 prusa slicer will hang. the status bar reads "infilling layers"
download Hex_Chain_142x163_Assy.stl from [
](https://www.thingiverse.com/thing:5419104/files)

open prusaslider 2.5.0 
drag & drop Hex_Chain_142x163_Assy.stl onto prusaslider, 
click the "slice now button" 
[ThisWillHangPRUSASLICER.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9630137/ThisWillHangPRUSASLICER.3mf.zip)

progress will get to 70% and hang, and after a couple min prusaslicer will become unresponsive (Turns red in the Mac activity monitor and has to be forcibly quit from the activity monitor or taskbar)

### Project file & How to reproduce

unzip the included prusa slicer project file 
OR
download Hex_Chain_142x163_Assy.stl from [
](https://www.thingiverse.com/thing:5419104/files)
THen
open prusaslider 2.5.0 
Open the uncompressed prusa slider file
OR
drag & drop Hex_Chain_142x163_Assy.stl onto prusaslider, 
then
click the "slice now button" 
and if you have a debugging tool you can watch prusa slider apparently wonder off into an infinite loop

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0-

### Operating system

macos 11.6

### Printer model

prusa mini

Lupo-Grigio, 2022-09-23:
this is the FIRST issue I've had with prusa slider in... um.. 3 years or thereabouts.  and I love my prusa mini. It has been rock solid for over a year. I wish I could afford a Mk3 (like the donated ones I built for the portland hackerspace). Keep up the good work! 

Oh, I also tried simplifying the model.. same result. hang....

kubispe1, 2022-09-26:
I tried it on Win 10, and it works to me. But to be honest, it took quite a while.
[image]

DragonFinlay, 2022-09-30:
Sliced fine for me too, after taking a while at 70% (infilling layers).
Parts like these are always taking quite a long time to slice since there are a lot of quick, short lines and retractions.
This is a bit inconvenient if you want to change a few settings and see the result in the g-code preview, but there's no way around it.

If you can, try only taking one or two parts and assembling the hex chain using instances. That way, PrusaSlicer only has to calculate these 1-2 parts.

I'll close this since PrusaSlicer doesn't actually crash and it's not a bug.

closed 2022-09-30 by DragonFinlay

FidelCapo, 2022-10-03:
Thank you for the report.
The calculation method for this infill has been slightly modified. In the next version (probably 2.6.0) the slicing of this object should be much faster.