prusa3d/PrusaSlicer #8933: Pressure equilizer in spiral vase mode

NHaag87, 2022-09-23:
### Description of the bug

While printing in spiral vase mode with pressure equlization enabled the printer will become infinetly slow at the end of the last bottom layer. This is due to the fact that there is a stop in printing when changing from the bottom layer to the actual vase outer contours. Pressure equilization then ramps down the speed to 0, which can become really slow depending on the maximum slope set.

[image]
[image]

It's not a severe bug as the printing is only unnecessarily slower. Printing will continue as usual. However, should be easy to fix.

The maximum slope value I used was 5 mm³/s².

### Project file & How to reproduce

1. Create an object in spiral vase mode printing
2. Set maximum volumetric speed slope to a value different than 0 (e.g. 5)
3. Extreme slowdown occurs when finishing the last bottom layer (default 3)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 10

### Printer model

Prusa Mini+

NHaag87, 2022-11-06:
> Can you provide us your .3mf? Could be nice to observe it on this specific example. Thank you for your time.

Yes, sure. Here you go:
[HexSurfaceVase5_Example_PE.zip](https://github.com/prusa3d/PrusaSlicer/files/9946213/HexSurfaceVase5_Example_PE.zip)

Sorry for the late reply btw.

kubispe1, 2022-11-09:
Thanks… But I see now it is not only problem of your specific model, but general behavior. 
In the end of bottom infill extruding needs to slow down to ~ 0. Than on beginning of vase procedure it starts from ~ 0. Picture shows extruding for settings without pressure equalizer. 
So I think, when you apply pressure equalizer you extend slowing down and starting up much more.
[image]

awabom, 2022-11-21:
Encountered this bug today (in 2.5.0) - maybe it should not be tagged as "waiting response" anymore?

closed 2026-09-29 by prusaslicer-bot[bot]