prusa3d/PrusaSlicer #8937: Arachne Does Not Obey Slicing Resolution

ruredi87, 2022-09-24:
### Description of the bug

I noticed arachne causes some misalligment at walls, specialy on curved walls. If you look the pictures there is artifacts on outer walls not because of variable line width, because of generated printing paths and their random allignment. Also arachne generates walls with much less gcodes compared to classic slicing engine.  I randomly picked a layer and analyzed it. With classic slicing mode the outer wall complating with 163 commands (between 174176-174339 lines). But with arachne it is only 34 gcode commands (between 60023-60057 lines)

Arachne's this behavior causes artifacts on walls and much less resolution on walls. plesase see the pictures. There is no problem with classic slicing engine.
[image]
[image]

### Project file & How to reproduce

[bmg project (2).zip](https://github.com/prusa3d/PrusaSlicer/files/9639692/bmg.project.2.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Build: PrusaSlicer-2.5.0+win64-202209060714

### Operating system

Windows 11

### Printer model

Ender 3 v2

kubispe1, 2022-09-26:
Hi, thanks for report. We detected this problem yet. I think this is duplicate of #8524, so I am closing it. 
I also recommend to check this comment: https://github.com/prusa3d/PrusaSlicer/issues/8524#issuecomment-1255047681. Quite nice description (Increase of resolution of exported .stl is needed).

hejllukas, 2022-09-26:
I reopen this issue. This is a different issue than #8524.
This issue is caused by preprocessing of input polygons for Arachne and post-processing of perimeters generated by Arachne, which doesn't respect the value of slicing resolution.

DragonFinlay, 2022-10-04:
It is however the same issue or at least very similar to #8700.
@hejllukas Enough to close this one again?

TalalMash, 2022-11-27:
@DragonFinlay Same issue here, OBJ file with very fine resolution, completely fine with Classic:
Classic:
[image]
Arachne:
[image]

TalalMash, 2022-11-27:
*Thank you for your efforts!

Mark-Dunn-OIL, 2023-02-13:
I have, for some time, been trying to understand where the resolution of my printed gears have been removed from my design. After reading this I switched between the two algorithms. For my gear teeth:
Arachne issues 7 gcodes
Classic issues 39 gcodes
I realise that in this case 39 gcodes may be too many, but I would like it to be under my control as to how much approximation (deviation) I am prepared to accept.

Is it too much to ask for the benefits of arachne with the resolution of classic?

rongith, 2023-03-22:
Indeed, Arachne produces too much small lines regardless of the specified gcode_resolution parameter.
Here the figures:
<table>
<tr>
	<td>Perimeter generator
	<td>G-code resolution
	<td>movement lines (figures taken from the preview window)
<tr>
	<td>Arachne
	<td>0.0125
	<td>320976
<tr>
	<td>Arachne
	<td>0.0625
	<td>318325
<tr>
	<td>Classic
	<td>0.0125
	<td>223783
<tr>
	<td>Classic
	<td>0.0625
	<td>104567
</table>

Those multiple short lines produce zips, blobs and the serial connection from octoprint struggle. There is a lot of pauses during the printing waiting for the commands to arrive to the printer.

With the Arachne generator I didn't find a way to reduce the number of movements to fight issues described here: https://www.cnckitchen.com/blog/removing-zits-and-blobs-on-your-3d-prints-adjust-your-slicing-resolution

[TestLattice-r01a - Contenant.zip](https://github.com/prusa3d/PrusaSlicer/files/11042085/TestLattice-r01a.-.Contenant.zip)

rongith, 2026-09-07:
Still an issue.