prusa3d/PrusaSlicer #8938: No toolbar icons on Linux

rcmz, 2022-09-25:
### Description of the bug

I just noticed that the Windows version of PrusaSlicer includes icons for the toolbar as well as for the Simple/Advanced/Expert buttons, yet the Linux one does not :
[image]
[image]
I think those icons look really nice and help readability so is there a reason to not include them on Linux ?

### Project file & How to reproduce

Nothing to say here

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.5.0+linux-x64-GTK3

### Operating system

Manjaro Linux

### Printer model

Doesn't matter

kubispe1, 2023-02-06:
Ok, sorry it is not fixed with 2.6.0. My mistake.

rcmz, 2023-02-06:
😢

rcmz, 2023-02-06:
sorry miss clicked didn't want to close this topic

kubispe1, 2023-02-06:
Can I ask you for extra info on what kind of desktop environment you use? I realize that it also could be an especially for your issue critical. And by the way Have you already tried 2.6.0-alpha3 ?

rcmz, 2023-02-07:
I'm using Gnome 43 with a color tweaked version of the default theme. I can confirm that the icons are still missing in the 2.6.0-alpha build. Also I know this is not the right place for this, but the issue of excessive infill near slanted walls is still present ! This is seriously my biggest gripe with prusa slicer and it has been for years.
https://github.com/prusa3d/PrusaSlicer/issues/9245
https://github.com/prusa3d/PrusaSlicer/issues/8727
https://github.com/prusa3d/PrusaSlicer/issues/6853
https://github.com/prusa3d/PrusaSlicer/issues/6513

closed 2026-09-29 by prusaslicer-bot[bot]