prusa3d/PrusaSlicer #8941: Incorrect nozzle temperature

drelich, 2022-09-25:
### Description of the bug

PS 2.5 (arm version for macOS) shows Temperature in the View Temperature as 0°C. 

On top of that, 'other layers' temperature is not applied in gcode. I have my filament profile set to 240°C for first layer and other layers to 245°C but the temperature does not change when printing.

### Project file & How to reproduce

[keybak_mount.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9641511/keybak_mount.3mf.zip)
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+MacOS-arm64

### Operating system

macOS 12.6 Monterey

### Printer model

EnderWire/Switchwire with Stealthburner

rtyr, 2022-09-26:
Reproduced. 0C for first layer is expected, because of start g-code, but the temperature for other layers should be 245C.

It seems that g-code preview does not process G10 command properly (used with RepRapFirmware g-code flavor). In this case `G10 S245 P0`. FYI @enricoturri1966

drelich, 2022-09-26:
Ah, okay. Seems kinda odd that the slicer's own Temperature Preview isn't aware of the temperatures right after it's done slicing. After all, it is passing the first layer temps to the START_PRINT script...

As for the G10 command, my MainSail terminal does throw an 'Unknown command:"G10"' with each print when switching to layer 2...

rtyr, 2022-09-26:
> As for the G10 command, my MainSail terminal does throw an 'Unknown command:"G10"' with each print when switching to layer 2...

That's because you are using incorrect g-code flavor (RepRapFirmware). RepRapFirmware uses G10 command by default. But since you mentioned Mainsail, it is now clear you are in fact using Klipper FW, not RRF. Use `Marlin (legacy)` g-code flavor for Klipper (as instructed on Klipper website). We are going to add separate Klipper g-code flavor in the future.

> Ah, okay. Seems kinda odd that the slicer's own Temperature Preview isn't aware of the temperatures right after it's done slicing. After all, it is passing the first layer temps to the START_PRINT script...

G-code preview uses temperatures from the G-code. So `M104 S0` in this case. We don't process that START_PRINT Klipper g-code macro (which can also use different name). We may improve the handling of the start g-code for future Klipper g-code flavor. You can use placeholder instead of S0.

The issue with G10 in the preview will be fixed.

drelich, 2022-09-26:
Good catch! Not sure how I managed to change the flavour! Always used Marlin 2 (or Marlin legacy), ffs. Thanks!

The code in Start G-code field of PS is

`START_PRINT BED_TEMP=[first_layer_bed_temperature] EXTRUDER_TEMP=[first_layer_temperature]`

That means, PS should be able to see the first layer temps just fine if it's able to pass them on? Or am I not understanding it correctly?

In any case, thanks for your support! Fixed my gcode flavour so that should at least take care of the 'other layers' temp for me :)

rtyr, 2022-09-26:
> Always used Marlin 2 (or Marlin legacy)

Do not use `Marlin 2` flavor, it is not good for Klipper, because it produces M204 commands in the format, which is not supported by Klipper.

> That means, PS should be able to see the first layer temps just fine if it's able to pass them on? Or am I not understanding it correctly?

G-code preview doesn't know such script. It currently supports only usual g-code commands such as M104/M109 or G10 for RRF. We could just use the temperature from the filament settings, but it would not always correspond with the actual g-code (your start g-code is a good example). As I mentioned, we plan to add separate Klipper g-code flavor.

drelich, 2022-09-26:
Thanks for the explanation. Looking forward to native Klipper support! :)

closed 2022-09-26 by drelich