prusa3d/PrusaSlicer #8943: UI: add "Fill pattern" ListBox on the Plater tab (i.e. just next to "Infill percentage" ListBox)

rsebestik, 2022-09-26:
I am always frustrated while setting infill pattern and percentage, that percentage i can set on Plater tab, while for pattern i must go on different tab (Print Setiings -> Infill).
Every pattern needs different percentage, so it is logical to set them together from the main tab. When experimenting which pattern suits the print the best - it is annoying to select bext combinations of pattern and % on different places.

**Describe the solution you'd like**
Just add iit please in the UI - also on the Plater tab, for example just next to Infill %.

[image]

foreachthing, 2022-09-27:
Similar to: https://github.com/prusa3d/PrusaSlicer/issues/8320

rsebestik, 2022-09-27:
well it is not that similar, I would like to have it on the main "Plater tab" - where you can see your 3D model not under another Print settings tab under new menu of favorites.
Just wanted to highlight the difference :)

foreachthing, 2022-09-28:
Right. I'll find (or take) a screenshot later, to illustrate what I meant. Slic3r has a favorites function, where those are listet where Prusa lists objects.

foreachthing, 2022-09-28:
So, here are some screenshots from Slic3r:
Setting shortcuts:
[image]

And the main gui changes:
[image]

t-makaro, 2023-09-11:
I was just looking to see if this was possible. This would be a great addition. Personally, infill pattern is one of my most changed settings.

closed 2026-09-29 by prusaslicer-bot[bot]