prusa3d/PrusaSlicer #8948: Autoarrange wont work if Sequential printing is enabled

aclucca, 2022-09-26:
### Description of the bug

If you autorotate this single object PS tells you that it is too large.
Turning off  "Sequential printing" fixes this behaviour.

### Project file & How to reproduce

[image]
[food_dispenser.zip](https://github.com/prusa3d/PrusaSlicer/files/9648023/food_dispenser.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Windows 10 64bit

### Printer model

Prusa MK3

kubispe1, 2022-09-27:
If you want to arrange a single object for sequence mode, you have to **(1.)** decrease the clearence, and then **(2.)** you can also decrease the sapcing in the arrangement option to fit the object. After this arrangement it works fine. This is a very specific use case.
**(1)**
[image]
**(2)**
[image]

aclucca, 2022-09-28:
thanks for the tips on how to solve / work around the problem . . .

I think it would be better, however, if there is ony one object, simply ignore these two settings, because they have no use with only one object.

lukasmatena, 2025-02-18:
Arrange and crash detection for sequential printing was improved in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1). I believe it solves this issue. Your feedback is welcome. Closing.

closed 2025-02-18 by lukasmatena