prusa3d/PrusaSlicer #8950: [RRF] No tool selected when GCODE start cause error when print start (td;dr; missing T0 at the beginning of GCODE)

noisemaker00, 2022-09-26:
### Description of the bug

The generated GCODE lacks of the tool selection command (eg. T0) before the `set extruder temperature` command
This will cause an error with RepRap Firmware:
`Error: Attempting to extrude with no tool selected.`

You can find a reference to this problem here too:
[https://forum.duet3d.com/topic/20810/problems-with-prusaslicer-2-3](https://forum.duet3d.com/topic/20810/problems-with-prusaslicer-2-3)

Slicing the same STL with eg Cura generate a .gcode with tool selection correctly added before any attemp to heat up the extruder

### Project file & How to reproduce

- select RepRap as firmware for the printer
- slice any STL
- open the .gcode file: the T0 command is missing before the `set temperature` command

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows

### Printer model

custom

rtyr, 2022-09-27:
I would say it is just about using correct start g-code.

You can for example use.

```
T[initial_tool]
G10 S[first_layer_temperature] P[initial_tool]
```

noisemaker00, 2022-09-28:
I think it's slicing software duty to select which tool to use to print.
This a sample part of a gcode file generated by PrusaSlicer:
```
...
M190 S80 ; set bed temperature and wait for it to be reached
G10 S236 P0 ; set temperature
;TYPE:Custom
<custom start gcode>
...
```
and this is the same 3D object sliced with Cura
```
...
;Generated with Cura_SteamEngine 4.12.1
T0
M190 S80
M104 S236
M109 S236
M82 ;absolute extrusion mode
<custom start gcode>
...
```

rtyr, 2022-09-28:
You can easily set it up that way as I mentioned. Anyway, I am closing this as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/1796.

closed 2022-09-28 by rtyr

noisemaker00, 2022-09-28:
the custom start gcode section will be added after the tool temperature initialization command as you can see in the above gcode, so the error is stil raised by RRF because the first attempt is to heat up something without selecting it.
I'll try the way you propose to solve by now, but is not a correct behaviour imho

rtyr, 2022-09-28:
> the custom start gcode section will be added after the tool temperature initialization command as you can see in the above gcode

No, it will be added in specified order as described in my comment. PrusaSlicer doesn't emit temperature commands automatically if those are already detected in custom section (as described in the tooltip). People are often using custom heating sequences (different order, partial heating for MBL, etc.).

[image]