prusa3d/PrusaSlicer #8951: Optimized sidebarsize after slicing (eg for small PC screen)

SirFlour, 2022-09-26:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

I use a small laptop and when i slice i dont see the printing time (Expert Mode). The sidebar gets smaller, but still i have to scroll down. 

[image]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Making the size of the Box, where the object name is smaler, as it is not needed so big. 

[image]

**Describe how it would work**
Be as descriptive as your skill level allows you.

rezisable or automatlically adjust to perfect size. it looks like, it gets smaller to a specific size after slicing.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

If too much work, or not important enough, i can live with it to scroll down :)

**Additional context**
Add any other context or paint drawing about the feature request here.

Many thanks and have a nice day

mdeeter, 2023-09-21:
I agree with this.... but I would add that making these panels collapsible would be a decent change as well.

[image]

I too use a 13" macbook and it's really annoying that I have to scroll down to see print time changes as I make changes to print options.

SirFlour, 2023-09-21:
that is a good point and hopefully make the enhancement easier. As i am testing with different setting the print time and fillament usages, i dont need to see the objects and so it can be collapsed :)

closed 2026-09-29 by prusaslicer-bot[bot]