prusa3d/PrusaSlicer #8953: Add two .3mf files, auto arrange makes one of them disappear!

cjshaker, 2022-09-27:
### Description of the bug

Add two .3mf files, auto arrange makes one of them disappear!
[Yeti_Mug_Holder_and_bracket_fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/9650694/Yeti_Mug_Holder_and_bracket_fixed.zip)
[image]
[image]
[image]
[PrusaSlicerBUG.zip](https://github.com/prusa3d/PrusaSlicer/files/9650712/PrusaSlicerBUG.zip)

### Project file & How to reproduce

Drag Bracket_fixed.3mf onto the platter (See Step1). Drag Yeti_Mug_Holder_fixed onto the platter. (See Step2).
Select 'Arrange'. The Yeti_Mug_Holder gets moved off the platter and disappears! (See Step3).

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 11 22H2  22621.607

### Printer model

Prusa MK3S+

cjshaker, 2022-09-27:
These files are from "Yeti Cup and Mug Holder by Reneb3"
www.thingiverse.com/thing:2749185
Chris

cjshaker, 2022-09-27:
Not sure if this is a problem with the mug STL file, or with Slicer, but you can see that it does NOT slice correctly. Drag the original file, Yeti_Mug_Holder, onto the platter (See Mug1). As soon as you attempt to slice it, it turns into a round disk on the platter. Very strange (See Mug 2).
[image]
[Mug_Slicer_BUG.zip](https://github.com/prusa3d/PrusaSlicer/files/9650948/Mug_Slicer_BUG.zip)

cjshaker, 2022-09-27:
[image]

cjshaker, 2022-09-27:
[image]

cjshaker, 2022-09-27:
I created Yeti_Mug_Holder_fixed.3mf from Yeti_Mug_Holder.stl with Microsoft's 3D Builder program, by opening the STL file, and then letting 3D Builder fix the problems it found.
[Yeti_Mug_Holder.zip](https://github.com/prusa3d/PrusaSlicer/files/9650961/Yeti_Mug_Holder.zip)
Chris

cjshaker, 2022-09-27:
I used Prusa Slicer 2.5.0 to "Repair STL file" on the Yeti_Mug_Holder.STL file, creating Yeti_Mug_Holder.OBJ. Cura 5.1.1 WAS able to slice that file correctly, but Prusa Slicer 2.5.0 was NOT able to slice the Yeti_Mug_Holder.OBJ file that it created. It failed and resulted in the round disk as in picture MUG2 above
[Yeti_Mug_Holder_OBJ.zip](https://github.com/prusa3d/PrusaSlicer/files/9651126/Yeti_Mug_Holder_OBJ.zip)

cjshaker, 2022-09-27:
Cura 5.1.1 IS able to slice the Yeti_Mug_Holder.STL file, but it does complain "The highlighted areas indicate either missing or extraneous surfaces. Fix your model and open it again into Cura". 
[image]

cjshaker, 2022-09-27:
Looks like a bug in Sketchup to me.

The author of Yeti Cup and Mug Holder updated his design since I first downloaded it. He included the sketchup source file.
https://www.thingiverse.com/thing:2749185

I opened his Yeti_Mug_Holder.skp in the free online version of Sketchup at https://app.sketchup.com/
I downloaded a New STL file, Yeti_Mug_Holder, and it still will NOT slice correctly. I was able to fix it using Microsoft's free 3DBuilder so that Prusa Slicer would slice it.

Attached "NEW_Yeti_Mug_Holder.zip" with the files I am talking about.
[NEW_Yeti_Mug_Holder.zip](https://github.com/prusa3d/PrusaSlicer/files/9656053/NEW_Yeti_Mug_Holder.zip)

murk-sy, 2022-09-28:
Both models actually appear to be completely screwed up. It's the reason why autoarrange doesn't work either - some of the vertices might as well be on the other side of the planet.

To make it at least printable, you can download v2s and use the cut tool to cut the bottom 0,6mm of the mug holder. Repair *after* the cut, not before.

cjshaker, 2022-09-28:
Thank you. I managed to make a good print of both parts on my MK3S+.
Chris

kubispe1, 2022-09-29:
I can confirm that atteached models are in really bad condition. Especially Yeti_Mug_Holder_fixed. There will be some improvements around model repair in next release, despite this Yeti_Mug_Holder_fixed does not look good, when I slice it...

cjshaker, 2022-09-30:
I verified that the bad files are created by Sketchup.
Chris

kubispe1, 2022-10-03:
Thanks for additional info. Seems to me it is clear. Can I close it ?

closed 2022-10-04 by kubispe1