prusa3d/PrusaSlicer #8954: .step filenames not fully displayed in object list when extra periods present

murk-sy, 2022-09-27:
### Description of the bug

After an object is inserted, the filename is cut off at the first period, regardless if it's the extension delimiter or not. Not very common, but can easily happen with company/customer names and dimensions in the filename.

Additionally, I would prefer to see .step extension visible - or, since it's converted to a mesh, call it .step* or something along those lines.

### Project file & How to reproduce

Create .stl and .step files with multiple periods in the filename and insert them into Prusaslicer.
[image]

Name of the .step file is cut off - the extension is not shown either, unlike .stl files.
[image]

Example model files
[box 10.0 mm.zip](https://github.com/prusa3d/PrusaSlicer/files/9657302/box.10.0.mm.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-win64

### Operating system

Windows 10 x64

### Printer model

Prusa Mini

rongith, 2022-10-11:
The same here.
[image]

rongith, 2025-10-28:
One first step would be to change in /src/libslic3r/Format/STEP.cpp:
`occt_object.object_name.erase(occt_object.object_name.find("."));`
to
`occt_object.object_name.erase(occt_object.object_name.rfind("."));`
in the function load_step

We would delete text after the last dot and not after the first dot.

For the moment I don't have time to fork, create a commit, compile, test and create a PR.

Here the explain of the original line:
```
[tamasmeszaros]   3 years ago (July 26th, 2022 2:48 PM) 

STEP: Implementation ported from BambuStudio:
CMake handling is different

STEP: Removed preprocessing stage

STEP: Small refactoring

STEP: Bigger refactoring

STEP: Changed naming on loaded object and volumes:
If the STEP contains exactly one named volume, the object and its first
volume will both have that name. Otherwise, filename w/o suffix is used
as object name and volumes are named using names from the STEP (if there
is none, untranslated "PartN" string is used).

STEP: Load the libraries dynamically on Win

wip
```
Then I need to dig a little bit more to see the consequences to keep the extension suffix as for stl files.

rongith, 2025-11-17:
The latest PR behavior:

[image]

rongith, 2025-11-17:
@kubispe1 , would you consider my PR to fix this issue?

rongith, 2026-09-02:
Still an issue in PrusaSlicer 3.0.0-alpha11.
Filename is cut after the first found dot.