prusa3d/PrusaSlicer #8958: bridge infill over an overhang doesnt meet the inner perimeter.

patrickmcquay, 2022-09-28:
### Description of the bug

Layer 63 of the project file, the bridge infill does not meet the perimeter. I would expect it to do so.

This appears to happen over perimeters that have overhangs underneath them.

### Project file & How to reproduce

[Doll House Door.zip](https://github.com/prusa3d/PrusaSlicer/files/9665490/Doll.House.Door.zip)
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

windows 10

### Printer model

ender 3, heavily modified, my own profiles.

kubispe1, 2022-09-29:
Hello, thank you for your issue. In my opinion, this is a duplicate of #8774
There are a bunch of problems with the bridges same like you have reported, hope we will improve it.
I am closing this issue with link...

closed 2022-09-29 by kubispe1

patrickmcquay, 2022-09-29:
One small note, this happens with infill on, as well as ensure vertical shell thickness does not affect it. I dont know if that makes it a separate bug or not.