prusa3d/PrusaSlicer #8959: Use pressure equalizer for sharp corners

vovodroid, 2022-09-28:
As it's known, sharp turns can create bulges and even bumps, and linear advance should resolved this.

Unfortunately not all printers/firmwares support LNA, so why not use pressure equalizer (PE) not only for feature changes, but also for sharp corners (e.g. placers with abrupt speed changes)?

**Describe how it would work**
Like PE slows/accelerates speed and flow rate toward moves to/from feature change it can do the same with corners.

[image]

n8bot, 2022-09-29:
Basically, slicer-based pressure/linear advance. I might be down with this.

Also could include variable retraction amount/speed based on travel distance. All while coordinating with the simulated motion planner to get accurate accel/decel profiles.

n8bot, 2022-09-30:
A thought occurred to me. This would be almost equivalent to manually decreasing the jerk and acceleration. If you have low jerk and low accel, every corner will be slowed down upon like you wish.

vovodroid, 2022-10-02:
> This would be almost equivalent to manually decreasing the jerk and acceleration

Besides reducing print speed.

n8bot, 2022-10-02:
Well, if you reduce jerk to something very low like 10 mm/s, then sharp corners like that will be performed at lower print speeds. If you reduce acceleration to something low like 200 mm/s^2, then the reduced corner speed will be more pronounced, and there will be a more gradual reduction in speed.

Since the E axis movement is coordinated with the X/Y axis, this will have the intended effect of reducing pressure in the nozzle.

closed 2026-09-29 by prusaslicer-bot[bot]