prusa3d/PrusaSlicer #8961: Cut a slice out feature

ColinWise, 2022-09-28:
A request for an addition to the CUT feature .
A second cut line to create a centre section.
The ability to then cut a slice out of a part and have the top section drop onto the top
of the bottom section with the centre section removed.

OR.. 
Allow Z to be moved in the + direction , so as a cut part can be placed on the top of another part.

For a simple example if you had a cylinder stood upright with a cross hole near the top 
and a cross hole near the bottom you could reduce the overall hight without losing the cross holes.

neophyl, 2022-09-29:
You can already do this.  Cut into the parts you want.  Then merge the bits you want into one Object.  You can then select the Parts using the tree view on the right and move them wherever you want, to place one on top of the other.
You can only select Objects on the plater, to select Part of an Object you must use the right hand panel.

ColinWise, 2022-09-29:
Thanks I've  missed the working with the merged parts with the tree.
Thanks
For your help.
Regards colin

kubispe1, 2022-09-30:
Thanks you both for discussion. I think that's clear and I am closing this issue...

closed 2022-09-30 by kubispe1