prusa3d/PrusaSlicer #8962: Overlapped objects on build plate don't slice properly

RidwanHughes, 2022-09-29:
### Description of the bug

I created a simple & effective TPU cable tie https://www.thingiverse.com/thing:5523089 where the end user can create their own custom ones simply by moving the parts round on the build plate so they overlap by 2mm and then slicing, but the resultant slice tells the printer to print more than once in the overlapped spots. 

[image]

The only way to get a proper slice is to export the build plate as an STL or merge the objects manually, meaning if you want to change their positions after slicing you need to split them and then merge them again manually to slice again.

[image]

I thought overlapped objects would be automatically merged as part of the slicing process.

### Project file & How to reproduce

Load up the Hoopin_TPU_Cable_Tie_test.3mf file in the zip, and press "Slice now".

[Hoopin_TPU_Cable_Tie_test.zip](https://github.com/prusa3d/PrusaSlicer/files/9670425/Hoopin_TPU_Cable_Tie_test.zip)

Now merge the 4 objects and slice again.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 7

### Printer model

Prusa i3 Mk3

neophyl, 2022-09-29:
Yes.  This is known operation.  There are many issues already around this, most of them closed.  The user is expected to check for this and merge them if required.  Same with support structures for different objects.

You can reposition merged objects (which then are classed as Parts of a single Object) but you have to select them using the tree view on the right as you cant click on parts via the plater.

kubispe1, 2022-09-29:
Hi, this is a logic of PS. At the moment there will not be improve in this way...

closed 2022-09-29 by kubispe1