prusa3d/PrusaSlicer #8963: Slicing Graphics error

jjr0060, 2022-09-29:
### Description of the bug

When I try and slice a project my screen keeps flashing black. My computer then pops up with a  notification that says Application has been blocked from accessing project hardware.

### Project file & How to reproduce

[B10423-1.zip](https://github.com/prusa3d/PrusaSlicer/files/9670548/B10423-1.zip)

All I did was add custom support and nonsupport drawings then sliced it. 

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5

### Operating system

windows 11

### Printer model

Creality ender 3

jjr0060, 2022-09-29:
When I try to go back to the platter tab while it is flashing black the problem stops. I then tried to change the custom supports and re-slice the part. When I re-slice it the window just turns white. 
[image]

kubispe1, 2022-09-29:
Hi, thanks for your report. I am not able to reproduce it. I have (W 10 and PS2.5). 
We can try to check your graphics settings but not sure if we will solve it. Can you sent sytem info to us ? Here is way how to export it.
[image]

closed 2026-09-29 by prusaslicer-bot[bot]