prusa3d/PrusaSlicer #8964: PS 2.5 Arachne engine + supports: inconsistent layer height gcode (interleaving supports & model layers)

rsurgiewicz, 2022-09-29:
### Description of the bug

**PrusaSlicer 2.5** on windows (doesn't matter) with **arachne** engine generates gcode that produces poor quality print in some cases.
In my case supports are printed with 0.25 layer height and model itself with 0.3.
Such behavior causes that printer is "interleaving" support layer and model layers (observed on gcode preview).
[image]
[image]

When switching to "classic" engine, generated gcode print is normal.
[image]

In the end the quality of the print is degradated 
Arachne:
[image]
Classic:
[image]

I know that the print on this particular photo differs in layer height, but problem that I have written about could be seen regardless of the print itself (in gcode preview)

### Project file & How to reproduce

Normal behavior observed in "model_classic_OK.3mf"
Incorrect behavior observed in "model_arachne_NOK.3mf"
[Arachne_vs_classic_PS_2.5_inconsistent_layers.zip](https://github.com/prusa3d/PrusaSlicer/files/9672676/Arachne_vs_classic_PS_2.5_inconsistent_layers.zip)

Steps to reproduce:
1. Open "model_arachne_NOK.3mf"
2. Slice & go to Preview
3. Scroll between layers and drag horizontal slider to see actual tool moves
4. Observe that supports and model layers are misaligned and printed with some space between. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 10

### Printer model

Original Prusa MK3S

rtyr, 2022-09-29:
The support layers are not synchronized with the object layers, it is normal and intended behaviour, no matter what slicing engine is used. Supports are using independent (and variable) layer height. 

This allows to follow the gap between the object and the support top surface precisely. Other slicers (slicers not based on original Slic3r) usually allow the gap only in multiples of object layers. It also allows the support layers to grow faster than the object layers. It may of course have also some downsides, related to stringing etc.

rtyr, 2022-09-29:
It is currently not possible to disable variable layer height for supports, only for Multi-material prints with 0 contact distance.

There is feature request for it https://github.com/prusa3d/PrusaSlicer/issues/553.

rsurgiewicz, 2022-09-29:
Ok I see, so then it's a matter of quality deterioration when using arachne.
I thought that the quality is affected by behavior with layers being interleaved.

rtyr, 2022-09-29:
> Ok I see, so then it's a matter of quality deterioration when using arachne.

The prints on the photo are printed with very different settings, did you try it with same settings, only with different slicing engine?

rsurgiewicz, 2022-09-29:
Printing right now, the only difference would be perimeter generation engine.
I'll post an update within 1.5h

rtyr, 2022-09-29:
I believe I see what is happening. It is mainly about different order of printing perimeters with arachne with this particular model. "External" visible perimeters of the tube are printed first after long travel between the two tubes.

rsurgiewicz, 2022-09-29:
I don't know how much difference does it cause but somehow with "classic" two perimeters are more tightly placed next to each other
classic
[image]
arachne
[image]

rtyr, 2022-09-29:
> I don't know how much difference does it cause but somehow with "classic" two perimeters are more tightly placed next to each other

That is expected and also one of the reasons why to use Arachne engine. The wall is too thin for 2 perimeters with standard extrusion width, classic slicing engine just overlapped them.

n8bot, 2022-09-30:
Can't you disable the variable layer height in supports by changing your printer settings to have min and max layer height == the fixed layer height? It doesn't seem like the user is using any of the variable layer height features.

closed 2026-09-29 by prusaslicer-bot[bot]