prusa3d/PrusaSlicer #8966: Fuzzy skin using smart fill

Jonny360, 2022-09-29:
Fuzzy skin has some issues with its current implementation. I'd love to be able to use the paint on supports-smart fill to select a face and then modify it for fuzzy skin on just that face. Currently when you use a custom modifier body it will create a fuzzy skin on both sides of the modifier body, almost like its a new model cut into your original model. A smart fill selection seems like it could work to make the fuzzy skin on the outside of the model only and would be super convenient.

kubispe1, 2022-09-30:
Hi, I think your feature request is duplicate of these. So let's check them:
#6988
#7433

DragonFinlay, 2022-10-04:
Yup, we've seen this reported a few times and we're aware of it. Hopefully we can improve modifiers in the future to exclude these inner perimeters.

Closing as duplicate.

closed 2022-10-04 by DragonFinlay