prusa3d/PrusaSlicer #8969: Only one of two identical overhengs has support

vovodroid, 2022-09-30:
### Description of the bug

Two identical holes, but only one got support structure. I would expect both of them will be handled in the same way.

[image]

### Project file & How to reproduce

[DirectBack-Mount.zip](https://github.com/prusa3d/PrusaSlicer/files/9684114/DirectBack-Mount.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Win10 x64

### Printer model

Biqu B1 SE

kubispe1, 2022-10-03:
Hello, I discovered that you apply - don't support bridges (Print Settings / support material / don't support bridges). After that holes are supported. At the same time there are settings and blocker, I don't understand, but never mind. 
[image]

vovodroid, 2022-10-03:
> you apply - don't support bridges ... there are settings and blocker

May be, but anyway why result is inconsistent? And support blocker is not applied to hole locations.

closed 2026-09-29 by prusaslicer-bot[bot]