prusa3d/PrusaSlicer #8972: Crash when slicing

napter, 2022-09-30:
### Description of the bug

Click slice on the attached 3mf and prusa slicer crashes

### Project file & How to reproduce

[crash 3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9687025/crash.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5

### Operating system

Windows 11

### Printer model

Prusa MK3S+

neophyl, 2022-09-30:
Yeah, does for me too on Arachne.  Switching to classic and it slices fine.

clivius, 2023-10-06:
PS 2.6.1 crashes with perimeter generation set to arachne when slicing the attached file.  Switching to classic, slices without issue.
[mk4_heatbead_strain_relief_60_btm.zip](https://github.com/prusa3d/PrusaSlicer/files/12833747/mk4_heatbead_strain_relief_60_btm.zip)

FidelCapo, 2023-10-09:
SPE-1958

FidelCapo, 2023-10-09:
I was able to reproduce the issue with 3mf from @napter  but I was not able to reproduce it with 3mf from @clivius . We'll look into it.

clivius, 2023-10-09:
@FidelCapo I was notified about the crash by a user on Printables and confirmed with my setup. Let me know if you need any logs and how to generate them.

closed 2026-09-29 by prusaslicer-bot[bot]