prusa3d/PrusaSlicer #8973: PrusaSlicer 2.5.0 crashes when slice now button pressed, 5% infill PLA

epleace, 2022-09-30:
### Description of the bug

PrusaSlicer 2.5.0 crashes when "slice now" button pressed, 5% infill PLA.
Attached is the project file.
Happens every time.

### Project file & How to reproduce

[keyer.zip](https://github.com/prusa3d/PrusaSlicer/files/9686949/keyer.zip)
If I leave at the default 15% infill it works fine.
Version 2.5.0+win64

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

win 10

### Printer model

prusa i3 mk3s+

neophyl, 2022-09-30:
Working fine for me on 5%, same ps version 2.5.0 and windows 10 OS too.

epleace, 2022-10-01:
Thanks for looking into it. This afternoon I've had a similar crash happen
a few times on different files - and I did get that file to work on 20%
fill.
So something funny is going on ... but at least there is a way forward.
I love the printer. Thanks for all you folks do.
Eric

epleace, 2022-10-04:
> Can't reproduce it either, so it seems unrelated to the 3D model.
> Please keep us updated in case you find out more about what makes the
> slicer crash.
> I'd suggest doing a clean uninstall and reinstall.

Thanks for the update.
I continue to experience crashes when I hit the "slice" button.
It happens on a freshly opened slicer, and only sometimes. When I've done
one successful slice I can go back to the one that crashed and it works
fine.
It frequently happens when slicing .stl files from OnShape.
I wish I could be more specific - intermittent problems are famously
difficult, sorry.
Best,
Eric

Eric Pleace, KO4ZSD
508.240.4488³22 yes
1220 NW 28th St,
Gainesville FL 32605

lukasmatena, 2024-02-19:
Do you still see the crashes as of 2.7.2-alpha2?

epleace, 2024-02-19:
Hi,
I have not seen that problem for a long time now.
best wishes,
Eric

lukasmatena, 2024-02-19:
Thanks for the quick reply. Let's work with the assuption that whatever was causing it was already fixed along with something else. Closing the issue.

closed 2024-02-19 by lukasmatena