prusa3d/PrusaSlicer #8974: [Feature Request] Double-Walled Spiral Vase Mode

degroof, 2022-09-30:
Intent is to create more robust prints while still benefiting from the features of spiral vase mode.

The process would work as follows:
1. the extruder lays down the inner perimeter, spiraling up 1/2 the layer height
2. without stopping, it crosses over to the outer perimeter (red)
3. it lays down the outer perimeter, spiraling up the remaining 1/2 layer height
4. again, without stopping, it crosses over to the inner perimeter (blue)
5. at this point, the extruder has increased a total of 1 layer in height, and the sequence repeats

At the crossover point, the cross-sectional area of the extrusion must be reduced, either by slowing the extrusion or by increasing the speed of the print head.

Some tuning of the algorithm would be required to hit the sweet spot between blobbing and gapping at the crossover point. 

As a proof of concept, I generated gcode to print a double-walled square print. I managed to get tune the print to nearly-negligible artifacts. (see attached)

[image]
[image]

DragonFinlay, 2022-10-04:
Interesting concept, although I don't see the real benefits from this.

The main selling points of the vase mode are being seamless, fast and easy to print.
Having two perimeters will introduce a seam again, and double the print time.

Feel free to tell me if I'm overlooking something, but personally I would rather switch to a bigger nozzle than using a two-perimeter vase mode.

dartrax, 2022-10-12:
Reminds me of #7456

mavelli, 2022-11-13:
interesting concept. while the wall thickness can be changed by using a bigger nozzle, there is a limit to that argument. material flow and nozzle compatibility. 
introducing the seam is not optimal... how about a 3 wall thickness vase mode where the spiral spirals back and forth?

mavelli, 2022-11-13:
[image]

DragonFinlay, 2022-11-15:
I'm not sure what that would be beneficial for as it introduces lots of seams all around the object? So the opposite of one of the main advantages of vase mode...

mavelli, 2022-11-15:
It still is a continous spiral without any crossing perimeters. So no weak points. It may be aesthetically not smooth with the lines introduced but it should not have any weak point seams.

dartrax, 2022-11-16:
> It still is a continous spiral without any crossing perimeters. So no weak points. It may be aesthetically not smooth with the lines introduced but it should not have any weak point seams.

Look at the one loop perimeter option in SuperSlicer. It satisfies all those arguments and on top of that tries to minimize the „looping seams“.

mavelli, 2022-11-21:
i checked out the one loop perimeter in superslicer, thank you. interesting, but no, it does make a seam with a weak point. instead of crossing, it loops, but it is each layer one seam. what i suggest is a continous spiral but extended to multiple thickness of extrusion. gradual increase in Z, no seams (folds are not seams in my understanding).

dartrax, 2026-09-18:
This is an interesting feature request, I think it should kept open for PrusaSlicer 3.