prusa3d/PrusaSlicer #8976: Increasing volumetric flow limit and movement speed does not improve print time

Riconec, 2022-10-01:
### Description of the bug

Running 2.5.0 release and I have plans to set up improved extruder with 30-40mm^3\s (mosquito magnum) instead of 20mm^3\s supercharged V6. When I increase volumetric flow limit and give more speed in sliced model I indeed see that model will be printed with 30mm^3\s but it gives little to no improvement for print time. 50% increase in flow gives like less than 10% print time improvement. 

I verified that flow increase and speed increase is there in sliced model, but if my model is printed with faster moves and at 50% higher flow rate why I get so small time improvement? I don't expect 50% time reduction, but at least 30% should be feasible because most of the model is printed at max volumetric flow. Cooling is 100% all the time, there are multiple copies of model printed together and cooling is out of equation. 

Since I don't have this mosquito setup printing yet I only rely on the slicer for the estimations. 
To rule out other machine limits I tried :
- increase retract\de-retract speeds and decrease retraction length
- increase head movement speed when not printing
- increase all accelerations a few times including extruder
This does improve speed a bit but still far from what I expect.

I see a few possible variants:
- Slicer does include some kind of slow down however this slow operation is not shown in feature view and view options in sliced model view
- Slicer have some limitations when it comes to time estimation on the high volumetric speed printing
- Slicer incorrectly shows print speeds (speeds are limited in gcode and time estimation but slicer shows full speed in sliced model preview)

I am happy to provide specific information but I need to know which exact one. On the test model I have:
- 10mm3\s flow - 5h print
- 20mm3\s flow - 3h print (2 times more flow, -40% print time)
- 30mm3\s flow - 2.5h (3 times the flow, -50% print time)
- 40mm3\s flow - 2h27m (4 times the flow, almost the same -50% print time)

### Project file & How to reproduce

[test.zip](https://github.com/prusa3d/PrusaSlicer/files/9689915/test.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

win10

### Printer model

ender 3 supercharged V6 but this only about slicer

rtyr, 2022-10-01:
Changing the volumetric limit will not change other (more) important factors. Like the fact, that you are printing 3 small objects with very small acceleration (500mms2). The printer will not be even able to accelerate to target speed in this scenario. PrusaSlicer emulates the firmware, so even time estimate is affected by it. 

I don't see any issue, so I am closing the issue.

closed 2022-10-01 by rtyr

Riconec, 2022-10-01:
@rtyr Have you read the ticket? I modified accelerations to absurdly high values (30000) not reachable on my machine and this does not provide improvement. If I use 3000mm\s^2 acceleration almost 90% of 30mm move is at full speed. Have you considered that?

rtyr, 2022-10-01:
> I modified accelerations to absurdly high values (30000) not reachable on my machine and this does not provide improvement.

You probably forgot about something.

Your project
[image]

Your project witch higher machine limits (same 30 vol limit).
[image]

Same as previous but with 15 vol limit.
[image]

rtyr, 2022-10-01:
> If I use 3000mm\s^2 acceleration almost 90% of 30mm move is at full speed.

In your provided project, your print acceleration is set to 500mm/s^2, not 3000mm/s^2..

Riconec, 2022-10-01:
can you show please acceleration settings you got in the screenshot? I really put all accelerations 3000, then 30000 and still had 2.5h

Riconec, 2022-10-01:
[test30k.zip](https://github.com/prusa3d/PrusaSlicer/files/9690059/test30k.zip)
this gives me 
[image]

rtyr, 2022-10-01:
Remove default 500mm/s^2 acceleration setting in your print profile.

Riconec, 2022-10-01:
[image]
this is what I have now. do you mean 500 is still somewhere I don't see?

rtyr, 2022-10-01:
Print profile, not printer profile :).

[image]

Riconec, 2022-10-01:
[image]
ah, I think you are talking about this one, right?

Riconec, 2022-10-01:
that's indeed was the issue. To be honest it is feels like a bit of a mess. I found M201 and M204 commands that configure accelerations but because of this "acceleration control" it emits additional M204.. 
This setting was here from the beginning in the profile so went under the radar for me. thanks for help