prusa3d/PrusaSlicer #8977: PrusaSlicer 2.5.0 installer "not responding" hangs.

mavu, 2022-10-01:
### Description of the bug

Starting installer for PrusaSlicer2.5.0 on Windows 10
[image]

Stays like this for at least 15 minutes. nothing happens.

### Project file & How to reproduce

Download installer.
Start installer.
Observe problem.
Wait 15 minutes.
Kill installer.
Try again. 
repeat.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.5.0

### Operating system

Windows 10

### Printer model

Prusa MK3s

DragonFinlay, 2022-10-05:
Can you give us a bit more info please? Please include the specific installer your using and more information about your system.
I cannot reproduce this, the installer (both standalone as well as full package) work fine and don't get stuck.

mavu, 2023-10-19:
I'm sorry, i completely forgot about this.
I solved the problem by downloading the complete 500mb or so package that does not try to download anything and could install that.

I'm still having the problem with installing the current 2.6.1 version, but unfortunately I can't find the large package anymore. 
This is the download page for Prusaslicer: https://www.prusa3d.com/page/prusaslicer_424/
there are 2 buttons. a large download button, which leads to this url: 
https://cdn.prusa3d.com/downloads/drivers/prusa3d_win_2_6_1.exe

That is a ~80mb file, with what I assume is the online installer. this hangs like described above.

Then there is the link "Full Package (All Drivers & Apps)" which leads to this url: 
https://cdn.prusa3d.com/downloads/drivers/prusa3d_win_2_6_1.exe

This is also a ~80mb file. This link used to lead to a 500mb file (i'll call it offline installer) which i could run and install on my PC.

Could you try if you can run the installer if you don't have network connection?

lukasmatena, 2024-02-19:
This should be solved now. Closing.

closed 2024-02-19 by lukasmatena