prusa3d/PrusaSlicer #8979: Wider range of extrusion width with Arachne

espr14, 2022-10-01:
**Is your feature request related to a problem? Please describe.**
E3D V6 0.4 mm nozzle has flat surface of over 1 mm in diameter so it can print extrusions from 0.4 to 1 mm without any issue. Wide extrusion brings higher speed, strength and better adhesion (1st layer) but it cannot print fine details. Currently, user has to use fixed extrusion width for a specific type of extrusion and/or layer.

**Describe the solution you'd like**
Arachne perimeter generator can change extrusion width. So it could go from 1 mm to 0.4 mm and back automatically.

**Describe how it would work**
User would set extrusion width as a maximum extrusion width. Minimum extrusion width would be the nozzle diameter.

**Describe alternatives you've considered**
- Print everything in narrow extrusion - slow & weaker.
- Print everything in wide extrusion - miss fine details.

**Additional context**

[arachne.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9690969/arachne.3mf.zip)
### Wide
[image]
### Narrow
[image]

yoshimitsuspeed, 2022-10-03:
If I am understanding you correctly I believe you could essentially accomplish this already. 
You can set your desired extrusion width in advanced settings. So you could set them to say 1mm. Then at the very bottom of advanced settings you can change how narrow arachne can go. So you could set your minimum perimeter width to 0.4 or whatever you want. 

With that said running 1mm out of a 0.4 nozzle is probably pushing things a little too far. Admittedly you can do it within certain parameters if you run slow enough but you would probably get a better overall and more consistent print if you went something like 0.065 max. 
 Similarly now with Arachne you can go narrower than the nozzle width so if you are happy with o.4mm min width you could run a 0.6mm nozzle with 1mm max and 0.4mm min.

espr14, 2022-10-04:
None arachne parameter makes a change. It sticks to extrusion width all the time.

Tupson444, 2022-10-08:
I tried tweaking some settings, and it seems that increasing "Perimeter transitioning threshold angle" solves this problem. Try increasing it to at least 30°.

(Print Settings > Advanced, under Arachne perimeter generator)

[image]

closed 2022-10-08 by espr14