prusa3d/PrusaSlicer #8980: Adding skin expand distance, horizontal expansion [Feature Request]

laptop-laptop, 2022-10-01:
**Is your feature request related to a problem? Please describe.**

Undesirable Benchy lines

**Describe the solution you'd like**
Somebody mentioned that using skin expand distance, horizontal expansion (not sure if they are the same) in Cura could eliminate Benchy line.

**Describe alternatives you've considered**
Add those feature(s) or whatever makes those ugly benchy lines disappear or less visible.

kubispe1, 2022-10-03:
Let's discover it better. I see, it calls **skin expand distance** as you mentiond. Seems to me, that Ps has a hardcode value?.!..

laptop-laptop, 2022-11-03:
Is it possible to make it user modifiable?

ismaildakrory, 2025-03-08:
I know this one is old, but it's 2025 and I couldn't find it in the slicer also.

closed 2026-09-29 by prusaslicer-bot[bot]