prusa3d/PrusaSlicer #8982: Raft layer height with support interface doubles max layer height

yoshimitsuspeed, 2022-10-01:
### Description of the bug

With settings 
Layer height 0.2
First layer height 0.15
Extruder min layer height 0.1mm, max layer height 0.22mm

With 1 raft layer support interface prints on the bed at proper height. 
With 2 raft layers it prints 1 full raft layer, then several layers before doing an interface layer at 0.45mm. With a dissolvable support this layer does not adhere well to the raft layer. 

If I increase number of raft layers it will increase the number of these 0.45mm layer height support interface layers. 
So if I set 4 raft layers it will print 3 interface layers at 0.45mm. 

Once it gets above those raft layers support interface prints at acceptable layer height within min/max window. 

### Project file & How to reproduce

With settings 
Layer height 0.2
First layer height 0.15
Extruder min layer height 0.1mm, max layer height 0.22mm

With 1 raft layer support interface prints on the bed at proper height. 
With 2 raft layers it prints 1 full raft layer, then several layers before doing an interface layer at 0.45mm. With a dissolvable support this layer does not adhere well to the raft layer. 

If I increase number of raft layers it will increase the number of these 0.45mm layer height support interface layers. 
So if I set 4 raft layers it will print 3 interface layers at 0.45mm. 

Once it gets above those raft layers support interface prints at acceptable layer height within min/max window. 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Win 10

### Printer model

Flashforge CP 2 with duet board.

DragonFinlay, 2022-10-05:
I can't reproduce it. For me, with 2 raft layers, it doesn't print several layers before the interface layer at 0.45, it only prints the interface layer at 0.45, as expected. Same for 4 raft layers - one full layer, one support layer, two interface layers.

[image]

yoshimitsuspeed, 2022-10-05:
@DragonFinlay 
Why would it print anything at 0.45 if layer height and max height are half that? That layer height really reduces the ability for PVA to stick and increases the odds of it not sticking at all. 

I did notice that it also seems pretty under extruded. Even at 30mm/s so I am confident it's not a volumetric flow issue. It almost seems like it is extruding the right amount for half that distance almost as if it is calculating for the right height but for some reason doubling the actual height or something. 

Either way why would it be doing a 0.45mm layer height if there is no setting telling it to do so and max layer height is set at half that?

DragonFinlay, 2022-10-06:
What you're saying is exactly right. Even though the g-code preview displays thicker lines, the extruded material is actually just corresponding to a normal layer, just at a higher than just one layer. With most materials, this is very beneficial because it makes the raft more "brittle" and therefore much easier to remove.

So the actual issue is the fact that you're having bad adhesion because the PVA doesn't stick due to the increased distance between these layers.
Try increasing the extrusion width for support material back to 0.45, or perhaps even beyond to something like 0.6 so there's much more PVA extruded. This should hopefully help. Feel free to keep me up to date.

yoshimitsuspeed, 2022-10-06:
@DragonFinlay 

Maybe there has been a miscommunication. I am talking about layer height, and looking back maybe you are talking about layer width?

DragonFinlay, 2022-10-07:
I'm also talking about layer height, or to be more precise, the height of the nozzle in Z.

Raft layer 1 - layer height (aka material extruded for height) 0.15mm - nozzle position 0.15mm
Raft layer 2 - layer height 0.2mm - nozzle position 0.45mm
Raft layer 3 - layer height 0.2mm - nozzle position 0.75mm
Raft layer 4 - layer height 0.2mm - nozzle position 1.05mm
Object layer 1 - layer height 0.2mm - nozzle position 1.45mm
Object layer 2 - layer height 0.2mm - nozzle position 1.65mm

The layer height numbers are just approximations, but the concept should be clear. On raft layers 2 to 4, the nozzle only extrudes material for 0.2mm layers while still moving 0.3mm higher, so the filament is laid down with much less adhesion to the previous layer. As mentioned, this is usually beneficial. If you now crank up the extrusion width for supports (which includes the raft) you should be able to make it work.

yoshimitsuspeed, 2022-10-10:
@DragonFinlay 

So the issue is with support interface layers if the support interface is building directly on the raft. 
If this explanation doesn't get us there I can reproduce it and upload a file but it has done it on every model I have tried so It should do it on any model. 

So let's say with a raft layer height of 2 because if I do raft layer height 1 it puts the support interface on the bed which I don't want. So with raft layers 2 it puts the support interface on top of the first layer. Maybe this is actually the bug as I would assume the first support interface layer would go on top of the second raft layer but it goes on top of the first. 
So it goes like this. 
Raft layer 1 - layer height 0.15 nozzle position 0.15
Then honestly to the best of my memory I actually think it starts laying down support grid layers. Not an actual second raft layer. So 
Support grid layer height 0.22  nozzle position 0.37
Support grid layer height  0.22 nozzle position 0.59
First support interface layer 0.45 nozzle position 0.59 

So to be clear it is printing one support grid layer without any raft or support interface. 
Then when it tries to lay the support interface down it is putting it on the first raft layer with a layer height of 0.45mm.

DragonFinlay, 2022-10-12:
Sorry, I don't think I fully understand your comment.

Raft is generated via the following logic:

1. A support interface layer is generated (for the model to sit on).
2. If another raft layer is needed, add a solid layer below (to increase adhesion to the print bed).
3. If another raft layer is needed, add another interface layer (to increase chances for a successful raft).
4. If another raft layer is needed, add a spaced support layer (same as above, but saving filament).
5. From here, if more raft layers are needed, there's always one solid layer at the bottom, two interface layers at the top and the rest is split 50/50 into interface and spaced support layers.

yoshimitsuspeed, 2022-10-13:
@DragonFinlay 
Hopefully this helps clarify the issue. 

[image]

yoshimitsuspeed, 2022-10-13:
I just tried to upload the PS file but it won't let me upload .3mf lol.

DragonFinlay, 2022-10-15:
I think what you mean is something I mentioned earlier. The yellow line is displayed as a thick line so it visually reaches the previous layer, but this is not representative of the amount of material extruded.

I'd still like to double-check your 3MF. Please ZIP it (or simply rename the ending from .3mf to .zip) to be able to upload it here.

closed 2026-09-29 by prusaslicer-bot[bot]