prusa3d/PrusaSlicer #8987: First layer on infill collides with nozzle because of curling up filament.

masterkyle, 2022-10-02:
### Description of the bug

When the first layer is printed onto the infill, it starts to curl up on the turnaround points. Because there is nothing where the filament can adhere on.
I think the problem can be solved when a supporting wall is already printed within the infill from the beginning. What also can be a solution, when this layer is printed all the way up to the outer walls of the part, so it can adhere there.

I added a screenshot of cura, where you can see, it first adds a circle of filament around the layer, so it can stick on.

[image]
[image]
[image]

You can find the part on thingiverse: https://www.thingiverse.com/thing:4591435/files
PlantPot_DiamondCut.STL

### Project file & How to reproduce

[PlantPot_DiamondCut.zip](https://github.com/prusa3d/PrusaSlicer/files/9692073/PlantPot_DiamondCut.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.4.2+win64

### Operating system

Win10

### Printer model

MK3S

rtyr, 2022-10-03:
I am closing this as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/569. This has quite a big priority.

BTW. The example from Cura would not help much (you can't print circles in the air).

closed 2022-10-03 by rtyr

masterkyle, 2022-10-09:
I printed the cura sliced gcode and it seems to work out. The ends of the first layer are connecting to the circle. Please see the picture: 

[image]

rtyr, 2022-10-09:
That is much more dense infill than on your initial images, I agree it helps in this particular case. You can see that "circle" lines are straight between ifnill lines. But in general, proper solution is https://github.com/prusa3d/PrusaSlicer/issues/569.

masterkyle, 2022-10-09:
You are right, the infill density is higher in the last picture. But i had this problems on even higher infill numbers without this circle. And yes, the "circle" is a polygon. Like you already said, it is impossible to print curved bridges. 
#569 doesn´t contain a real solution in my eyes. We will see what the future brings :)

rtyr, 2022-10-09:
With 569, bridges would be always anchored, so it should be the best solution.

masterkyle, 2022-10-09:
sorry for my dumb question, but how do I achieve this in my example print? Do you mean setting it to 100 % infill?

masterkyle, 2022-10-22:
rtyr said, with #569 bridges would be anchored, and called it the best solution. But I don´t understand. I see no solution in #569. In my eyes it is still an issue without any official solution?
So I asked, what I have to do, to achieve a print without the discussed problems?

DragonFinlay, 2022-10-27:
I think this is merely a misunderstanding. #569 will provide a solution to this problem once it is implemented.
For now there is no perfect solution to this, so the only thing you can do is to tinker with the infill (increase percentage, use adaptive infill, use modifier meshes, etc).