prusa3d/PrusaSlicer #8988: PrusaSlicer - add a preference to disable tooltips

markus-i, 2022-10-02:
More of an annoyance than a bug: some of the tooltips for modifer otions more or less completely cover up the option I actually wanted to change. Here's an extreme example:
[image]

If I only have one modifier, it might be tolerable - but with 20 modifiers (and counting) in one model, this gets rather annoying.

So, please: add a preference to turn off tooltips. Make it active per default and hide it in the Expert options, if you want - but let me have the option to remove them.

foreachthing, 2022-10-02:
I, too, would like to be able to disable tooltips. But it would be handy, if I could press a key (i.e. Shift) and the tooltip is shown. That way I could have both at the same time. Thanks!

NikosSprocket, 2024-12-22:
"..... be able to disable tooltips. But it would be handy, if I could press a key (i.e. Shift) and the tooltip is shown.
....."

That will be a great option, but at least be able to turn them off the tooltips popup that most of the time if covers the g-code window.

Skirge001, 2025-05-30:
**OMG, yesssss!!!**  I've been tweaking per filament g-code settings for the past few weeks and the popups are driving me **insane**!  Highlighting text, copying, pasting, reading, positioning the cursor, moving around, even freaking trying to **SCROLL** has the huge tooltip appear to cover the slider!  Give us SOME sort of option to disable these popups.  I even started looking to find a way to entirely disable all tooltips in Windows because I was getting so frustrated!  I see this has been open for nearly 3 years, so the chances are likely nonexistent, but I wanted to at least voice my disdain for this annoyance.

closed 2026-09-29 by prusaslicer-bot[bot]