prusa3d/PrusaSlicer #8990: original coordinates seem to be forgotten after closing and reopening a project (.3mf)

markus-i, 2022-10-02:
### Description of the bug

Couldn't find the issue already - but I'm afraid my ususal google-fu seemed at a loss as to how describe the search terms properly. _Someone_ surely has seen and reported this before?

So, what happened: 
- I had a project (.3mf) with one body and ~a dozen modifiers. 
- I had to restart the PC (project was saved). 
- Everything was ok, body and all modifiers in their expected positions and configuration
What I tried to do:
- create a new modifier in F360 and add it to the main body in the .3mf
What happened:
- modifier was added, but completely out of position with relation to the body

My suspicion: as long as I am working with the .3mf after importing the original body, the transformation how to get the body into its position on the printing plate is kept in memory and applied directly to any modifiers I add to the body. Once I close the .3mf (and/or when I open it again - but closing seems more probable) this transform is remved and only the coordinates relative to the printing plate are retained. Any new bodies/modifiers added afterwards are added with the (in my case pretty much arbitrary) offset between the body in f3d coordinates andd the body on the printer plate.

Background: the body is for a speaker box for a car (Citroen Ami/Opel Rocks-E), which is modelled from a scan of the dashboard to fir as good as possible intl the available space. Therefore, the absolute position and orientation of the body is pretty much arbitrary in F360's coordinate system.

Maybe a few pictures will help:
This is what it looks (with an easily generated test modifier) in F360:

[image]

This is what I expect to see in PrusaSlicer (using a .3mf that I never closed and re-opened):

[image]

And this is what I see if I import the same modifier (same .stl) into a stored and re-opened version of the  same .3mf:

[image]

If you attempt to repro, you of course won't be able to get that modifier into the correct position, since you'll be forced to work from a stored .3mf...

### Project file & How to reproduce

.3mf file:
[links_test.zip](https://github.com/prusa3d/PrusaSlicer/files/9692327/links_test.zip)

the stl for the additional modifier:
[test.zip](https://github.com/prusa3d/PrusaSlicer/files/9692328/test.zip)

How to reproduce:
You'll probably have to start with your own F360 project, as I can't send you the "virgin" .3mf from my PC...
1) create a F360 project with some sort of body at an arbitrary location and orientation, not on the coordinate origin
2) create a few more bodies relative to the first body (again, without paying any respect to F360's coordinate system), possibly intersecting that body (i.e. useful as modifiers - might also work for MMU designs)
3) import the original body (.stl) as body into Prusa Slicer
4) import the additional bodies (.stl's) as modifiers into Prusa Slicer
5) save the .3mf and close Prusa Slicer
6) create at least one more modifier body in F360 and export the .stl
7) Start Prusa Slicer and open the .3mf from the previous session
8) import the new modifier (from step 6)... and keep looking for it until you find it...

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Windows 10

### Printer model

Prusa i3 Mk 3 w/ MMU 2S

kubispe1, 2022-10-03:
I think you are totally right. Original coordinates system is ignored or lost. Only option that could be available is : ''Reload from disk (https://help.prusa3d.com/article/reload-from-disk_120427)
So you need to 1) update your .slt /. step - rewrite file !
than 2) reload from disk according help link
Please, can you try it and then write, if it suits of your workflow or if there are some issues or improvement, you would like to see.

markus-i, 2022-10-09:
> I think you are totally right. Original coordinates system is ignored or lost. Only option that could be available is : ''Reload from disk (https://help.prusa3d.com/article/reload-from-disk_120427) [...]

Sorry about the delay in responding, but I only got to do some f3d/printing on the weekend...
Ok, I think I tried what you were suggesting, but it didn't work out this time. I'll go through the various permutations of which .stl to update and reload once I find the time for it (that design is now pretty much done, so there's currently no pressure here - I'm just pouring the concrete into it, one layer of 250 grams per day...).

What I tried was:
- open the stored project in prusa slicer
- opened the project in f3d
- created a new test modifier in f3d
- exported that modifier in f3d, added it to the main part in prusa slicer -> still offset, as expected
- "reload from disc" the main part in prusa slicer -> quite a bit of activity (as expected, there are now 30 modifiers linked to it) -> no change
- added the new test modifier again to the main part -> same offset as before, so no change overall

kubispe1, 2022-11-11:
Hello @markus-i , I was not able to reproduce it last time. Now I find the path to reproduce it in PS 2.5.0 and also in recent release. There is probably some inconsistency in saving to 3mf. Conclusion:
- Reload from disc seems to me that works correctly
- But when you save 3mf close PS, than open and want to load new .stl (right click/Add Part/ Load), it does not respect relative position of other objects – **as you have described**

closed 2026-09-29 by prusaslicer-bot[bot]