prusa3d/PrusaSlicer #8991: Post-processing scripts error

zyd306, 2022-10-02:
### Description of the bug

https://github.com/mriscoc/Ender3V2S1/wiki/How-to-generate-a-gcode-preview

Cura 5.1.1 can be sliced and saved normally, but Prusa will report an error after slicing

### Project file & How to reproduce

[File.zip](https://github.com/prusa3d/PrusaSlicer/files/9692567/File.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5

### Operating system

Windows 10

### Printer model

Ender3 S1 Pro

Pejfa, 2022-10-03:
@zyd306  

Hello, why do you use a Post-processing scripts? In new release 2.5.0  you can chose the thumbnails format -> JPG  (_Printer Settings -> General -> Format of G-code thumbnails_)

zyd306, 2022-10-05:
https://github.com/mriscoc/Ender3V2S1/releases

I use this fireware,   the thumbnails format -> JPG (Printer Settings -> General -> Format of G-code thumbnails)  , the thumbnails can be generated, but cannot be correctly recognized by the device firmware

zyd306, 2022-10-05:
https://github.com/mriscoc/Ender3V2S1/wiki/How-to-generate-a-gcode-preview

Only by using that script can the generated file be previewed normally

zyd306, 2022-10-05:
Today I tested again. The preview generated by Pruda cannot be recognized correctly because the file code generated according to your settings is thumbnail_JPG at the start and end ,changing the start and end of the thumbnail can be correctly identified. However, it is slightly different from the script generated preview provided on the website. The script generated preview is better
[Preview.zip](https://github.com/prusa3d/PrusaSlicer/files/9716093/Preview.zip)

Jan-Soustruznik, 2022-10-06:
Hello, @zyd306 , 

there must be something wrong on your side ... I think. I tried to generate the thumbnail by script that you provided, check attached g-code: [Shape-Box_1h46m_0.08mm_210C_PLA_ENDER3.zip](https://github.com/prusa3d/PrusaSlicer/files/9728483/Shape-Box_1h46m_0.08mm_210C_PLA_ENDER3.zip)

Could you check if you have installed Pillow library and set correct  PATH to python in OS, please?

You can check by cmd, use the command below:
1. Python version: `python --version ` If you dont get the version, that means that the Python  isnt correctly integrated in your OS.
2. Pillow version:  `pip freeze` You show get the list of installed libraries where will be `Pillow==X.X.X`  If not, you have to install it. - https://pypi.org/project/Pillow/

zyd306, 2022-10-07:
I checked that Python and Pillow are installed correctly. I don't know if there is a problem with the path written in the post-processing script
[python.zip](https://github.com/prusa3d/PrusaSlicer/files/9735166/python.zip)

closed 2026-09-29 by prusaslicer-bot[bot]