prusa3d/PrusaSlicer #8992: Reverse print path if it's faster

ygoe, 2022-10-02:
**Is your feature request related to a problem? Please describe.**
This is a suggestion for an improvement. Sometimes, several parts of a layer are printed that are not connected with each other. Then it can happen that a middle part is printed in a direction that goes from the far end to the near end. So longer movements are needed between three parts.

Hard to explain in words. See the attached G-Code file (source 3mf file also included). Go to layer 8 (at 0.90 mm) and within that layer to the step 13305. You see the beginning of a fill pattern. From step 13346 on, the next area is filled, but from the far end to the near end. And finally from step 13478 on, the last area is filled, again at the far end.

**Describe the solution you'd like**
I suggest reversing the path of the middle part if the movements between the parts are then shorter. This should result in a reduction of the print duration, and thereby also reduction of power consumption (albeit minimal).

**Describe how it would work**
If the jump movements to a new part of the layer and then to the next part will be shorter if the next part is reversed, then do so.

**Describe alternatives you've considered**
Waiting a few seconds longer. Depending on the print geometries and number of copies, it could sum up to minutes.

📎 [Samples.zip](https://github.com/prusa3d/PrusaSlicer/files/9692712/Samples.zip)

murk-sy, 2022-10-04:
Note that reversing the direction can cause visible artifacts. An example is a shape like this, where on the inside there are 2 visible ridges at the edges of the holes. At the bottom, it's printing a donut, and at the ridge, it start printing multiple distinct islands. The outside has no artifacts, which is the desirable outcome.

[image]

For something like a draft/speed preset it could improve speed, but it should definitely not be the default (at least without changing the default direction behaviour).

ygoe, 2022-10-04:
Ah, I knew there was a catch. But is this relevant for infill? This generally shouldn't be visible after all.

BTW, I also noticed that the order in which independent parts of a layer are printed could also be improved. There's still a lot of unnecessary movement within a layer, back and forth, instead of from one end to the other (and then in the opposite order for the next layer). Not sure if that's related.

closed 2026-09-29 by prusaslicer-bot[bot]