prusa3d/PrusaSlicer #8993: Mirrored stl. Critical

BoevoyNarod, 2022-10-02:
### Description of the bug

After print finished I have acquired that the model was unexpectedly mirrored. I have check twice and also compared results with Simplify3D. It's critical bug due to hard to check on the fly like in my case such changes.  There is screenshot of Simplify3d where I load original STL and 3mf of Prusa slicer and screenshot of original model from CAD

### Project file & How to reproduce

Put attached STL to the table
[Filament Stand.zip](https://github.com/prusa3d/PrusaSlicer/files/9693393/Filament.Stand.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 10

### Printer model

It was born as Ender 3 pro

rtyr, 2022-10-03:
I don't see any issue. The object in your 3MF is just rotated 90 degrees to the left. Initial position after loading STL is the same as in the S3D.

[image]
[image]

If all prints come out mirrored on your printer (differently than the preview in the slicer), than that it is problem on the printer side (misconfigured firmware, XY direction, homing).

BoevoyNarod, 2022-10-03:
No, it is not rotated - see screenshot of original model then load STL (NOT 3MF - there already mirrored by PS) to a bed and carefully compare

BoevoyNarod, 2022-10-03:
My printer makes exactly what I see in PS. When I load STL
[image]
 and 3mf from PS to Simplify3d
[image]
 - I observe mirroring. 
[image]
And directly loaded STL to s3d is correct but 3mf from PS is mirrored from original model. It is not rotating Right and Left pieces could not be transformed each to other by rotation

rtyr, 2022-10-03:
Left, loaded STL, right loaded 3MF generated by PS.

[image]

rtyr, 2022-10-03:
Maybe also provide your S3D factory file. You may use some non-standard setting there (flipped axis).

BoevoyNarod, 2022-10-03:
I made a screen video
https://dropmefiles.com/atY8Q

BoevoyNarod, 2022-10-03:
I used S3D only to understand difference

bubnikv, 2022-10-03:
dear @BoevoyNarod 

While some of us developers have strong political views and opinions, we are keeping our github discussion strictly apolitical.
Therefore, your political view on war on Ukraine expressed by your nick name @BoevoyNarod (fighting nation, nation of worriers) is not welcomed here and your nick is banned from Prusa3D organization. If you want to continue discussion on this forum, please stop expressing your political views by your nick name or any other means. Thank you for understanding.

rtyr, 2022-10-03:
Not reproducible. The result is same with PS, S3D, Ideamaker. Closing.

closed 2022-10-03 by rtyr

BoevoyNarod, 2022-10-03:
Dear sir! My nickname is already 15 years old and me and my friend (trudovoy) made our e-mail addresses from domain name narod.ru as an alias of Yandex service. I have no any wish to demonstrate something to anybody. Don't judge others only from your own point of view. This is a technical forum and let it be clear of politics forever. Thank you.