prusa3d/PrusaSlicer #8994: First layer not slicing with entire model

minystreem, 2022-10-02:
### Description of the bug

When slicing my model only pat of the model is printed on the first layer, yes the model is perfectly flat. I expected all parts of the model to be on the first layer. I solution to this problem is to cut 0.01 mm off the bottom of the model. This problem is replicable on windows11 and windows10.

### Project file & How to reproduce

First Layer:
[image]
Second Layer:
[image]

[BatteryHolder2.zip](https://github.com/prusa3d/PrusaSlicer/files/9693757/BatteryHolder2.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

windows 11

### Printer model

Prusa Mini+

neophyl, 2022-10-03:
The model has 2 non manifold edges (which are not reported by PS).  When fixed (used Blenders 3dprint tool) and the fixed stl is loaded it is sliced properly.

Fixed file attached.
[BatteryHolder2_neo2.zip](https://github.com/prusa3d/PrusaSlicer/files/9696499/BatteryHolder2_neo2.zip)

minystreem, 2022-10-09:
ok

DragonFinlay, 2022-10-12:
Issue with the 3D model, closing as completed. Thanks @neophyl!

closed 2022-10-12 by DragonFinlay