prusa3d/PrusaSlicer #8998: STEP import excludes big chunk of part. STL import of same part slices as intended.

henriksundeodegard, 2022-10-03:
### Description of the bug

I've tried lately to use the new STEP import feature because it made the arrangement of this WIP print-in-place model a lot easier. However, the slicer seems to be struggling on this hinge.

Unsliced, PrusaSlicer warns that the STEP-file model has 127 open edges, while the same part imported as STL does not have any such warning. Also, when trying the option "Fix through the Netfabb" on the faulty part, it seems to fix the issue. But upon further inspection the hinge is not sliced correctly and would not work as a hinge..

Also asked this on Reddit: https://www.reddit.com/r/prusa3d/comments/xusdyq/prusaslicer_25_step_import_excludes_big_chunk_of/

### Project file & How to reproduce

[Relevant project files.zip (includes faulty STEP file and working STL file)](https://github.com/prusa3d/PrusaSlicer/files/9701317/Bug.report.STEP.import.not.behaving.zip)

---

### Screenshots

**1. Unsliced - PS 2.5 warns that the STEP file has 127 open edges:**
[image]

---

**2. Slicing as is makes this huge horizontal gap in this one part:**
[image]

---

**3. Tried the option "Fix through the Netfabb", but then the model is missing an indentation-feature on the hinge:**
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 11

### Printer model

Prusa i3 MK3S

n8bot, 2022-10-04:
Just curious, where did you export the STEP file? From what software? STEP files are not consistent. Even though there are standards for STEP files, there are many different standards they can follow and they all have errors sometimes lol.

henriksundeodegard, 2022-10-04:
I'm using Fusion 360 for designing.

n8bot, 2022-10-04:
Would you be willing/able to attach the fusion .s3d file? It could help in discovering what might be leading to the issue on import.

henriksundeodegard, 2022-10-04:
[Towel Paper Holder v65.zip - containing .f3d file](https://github.com/prusa3d/PrusaSlicer/files/9709491/Towel.Paper.Holder.v65.zip)

At this point I'll take all the help I can get.

n8bot, 2022-10-05:
I just had a chance to confirm that the fusion file doesn't have any problems with it. The export to step from fusion can be reimported into fusion and there are no errors.

When the step is brought into PrusaSlicer, the conical features and maybe others do not get imported. The conversion to mesh seems to be the problem.

henriksundeodegard, 2022-10-05:
That's great, thank you for finding a workaround.
Also didn't know about the multi-part import feature to keep the parts arranged as in the design in PrusaSlicer - will be useful for other projects as well. Thanks again!

By the way, is this a bug that might be long-term fixed in the future?

DragonFinlay, 2022-10-05:
I'd hope so! The STEP import is still very new to PrusaSlicer and bugs are to be expected.
This issue will be kept open along others of the same kind until we can get to fixing them.

bubnikv, 2022-10-05:
The STEP import quality depends on the OpenCASCADE CAD kernel
implementation quality. We cannot do much here at PrusaSlicer side, we will
have to wait for OpenCASCADE to improve.

It would be great if you guys could try to import into FreeCAD and if you
get the same issue, report it to OpenCASCADE folks. They may be thankful
for more input.

henriksundeodegard, 2022-10-05:
Seems to be only the inward conical features that's failing. I tried inverting the hinge "polarities" and this happened:

[image]

n8bot, 2022-10-05:
Somewhat surprisingly, the step files exported from Fusion 360 import fine into FreeCAD (the old version I had installed).

I do wonder if I remember seeing some commits to FC that were aimed at fixing step import. OCCT is a failed cad package from the 90s, at its core... as much as I love it... heh...
[image]

allesblinkt, 2022-12-20:
Same issue here with PrusaSlicer 2.5.0+win64. 
STEP file is exported from Fusion 360. It opens fine and is not broken in FreeCAD (0.20.1) and Rhinoceros (7.21). Currently my workaround is opening the STEP in Rhino and exporting it again.

[image]

Thank you very much though for implementing the STEP support. It is a major change in our printing workflow.
I have added one example STEP file which imports with missing faces in PrusaSlicer.

[C_DrivenPlate_Free_print_broken.zip](https://github.com/prusa3d/PrusaSlicer/files/10267556/C_DrivenPlate_Free_print_broken.zip)

jepler, 2022-12-30:
This printable has a step file that *looks* fine but slices wrong: https://www.printables.com/model/348322-odette-the-owl/files -- the stl file is fine. I think their workflow is fusion 360-based.
[image]
[image]

jepler, 2026-09-18:
The problem reproduces in whatever version was installed today with `flatpak install flathub-beta com.prusa3d.PrusaSlicer`

I imported the same "owl" model as in the comment above, using a fresh profile for a mk3s printer and the default step file import settings.

[image]