prusa3d/PrusaSlicer #9001: Incomplete Perimeter Extrusion

mv2woods, 2022-10-04:
### Description of the bug

Noticed that my prints were coming out messy around the seams. Almost seemed like the extruder wasn't starting the perimeters properly, like it was lagging the start. But after further inspection and watching the printer, I noticed it was actually the end of the perimeter that was not being printed. Like.. at all. I noticed that it was in series with a pretty inefficient move to hope to the other seam located on the opposite side of the circle being printed. I wondered if there was any connection with the 'avoid crossing perimeters' setting. So I disabled that and the prints are now looking normal, with complete perimeters.

The funny thing is I do not remember changing this setting recently, but I did just update to version 2.5.0.

I also checked to see if arachne vs classic changed anything. No change was noticed.

### Project file & How to reproduce

[image]
[image]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 10

### Printer model

Creality CR10 MAX

tjhalva, 2022-10-05:
This might be a similar issue to: https://github.com/prusa3d/PrusaSlicer/issues/8796

Does it always happen in the same location? It looks like there might be some layer shifting on the inside of the ring to the left of the vertical seam.

sakomoto987, 2023-07-04:
Did we find a solution for this?

mv2woods, 2023-07-04:
I just found a work around. Don't know if the actual bug was ever fixed.

closed 2023-07-04 by mv2woods