prusa3d/PrusaSlicer #9003: No 32bit build anymore?

richard9999999999, 2022-10-05:
Hello.
I am running 32 bit Windows7 machine in my workshop (the reason is I am using Mach3 CNC software there to control my CNC router and it doesn't run in 32 bit Windows). I use this machine also for PrusaSlicer. Please, could you consider still provide also 32 bit build? It  would help me much and I suppose I am not alone.
Thanks much,
Richard

kubispe1, 2022-10-06:
Hello @richard9999999999, you are right. We don't support 32 bits anymore and we don't plan to change it. This OS is in the end of support of others programs and applications. So, there are also vulnerable reasons.
Sorry for that complication

richard9999999999, 2022-10-06:
OK, thanks for the answer. I understand. Cold you then leave 32 bit configurations in the make files so potentially users could build its own 32 bit version by them?