prusa3d/PrusaSlicer #9004: Hangs on "Generating skirt and brim"

hub2, 2022-10-05:
### Description of the bug

## What is the problem?
Slicer is not able to finish the slicing in sensible amount of time (waited 10minutes+ and still no result).

## What did you expect?
Successfully created gcode file.

[image]
Usage:

[image]

### Project file & How to reproduce

https://github.com/prusa3d/PrusaSlicer/files/9714638/220110_filament-spool-holder_CR.zip

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 10 64bit

### Printer model

MK3S

kubispe1, 2022-10-06:
Somehow there is model defect, that PS doesnot detect.
[image]
When I fix model through the NetFabb it seems Ok and slice without problems.
In most recent developing version 2.6.0 alfa1 - it has still some troubles. 
@YuSanka , please can you check it ?

stjohnny22, 2023-01-22:
Here is another file having the same issue.  When brim is unchecked, it slices fine. I repaired through NetFabb which removed the multi-material painting, and all is well but once painting is reapplied, the hangup returns. I have loaded several other MMU prints and cannot replicate the issue with those but this one consistently does it on many different machines. Hopefully having this file to try and look at is useful in squashing this bug. 

[rebreather-box-lid.zip](https://github.com/prusa3d/PrusaSlicer/files/10473815/rebreather-box-lid.zip)

lukasmatena, 2024-02-19:
Fixed as of 2.7.2-alpha2. Closing.

closed 2024-02-19 by lukasmatena