prusa3d/PrusaSlicer #9005: Unmerging function missing

pavolNovak, 2022-10-05:
### Description of the bug

I merge two stls because I want the support to be generated nicely and not overlaping. BUt then I change something and I want to UNmerge them but there is no option for this wehn I right click. 

### Project file & How to reproduce

1. Get one stl and make a copy ( now you have two.
2. select both of them
3. right mouse button and select merge
4. now try to unmerge them ( you cant )
5. 
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-rc1+win64

### Operating system

Edition	Windows 10 Enterprise Version	20H2 Installed on	‎1. ‎12. ‎2021 OS build	19042.2006 Experience	Windows Feature Experience Pack 120.2212.4180.0

### Printer model

Prusa MK3S+

jmd73, 2022-10-05:
@pavolNovak You need to use the split function.

pavolNovak, 2022-10-06:
@jmd73 if its one model then it works yes. but if its multiple stl model then it brakes everything and you cant use it anymore because everything falls to the bed. And not always option to separate them by CUT function.  Its showing merged on top in model tree at right side so it knows. would be great to unmerge.

neophyl, 2022-10-06:
Split/Unmerge, same function different name.  If they are unmerged then they are separate Objects so the Objects rules applies and they all fall to bed. You couldn't have one off the bed before they were merged so what's the difference ?

pavolNovak, 2022-10-06:
@neophyl  If I open 3mf file that already contains merger models that happen to have modifiers and I want to reuse them. I cant even use the reload function because its not the same model So I need to UNmerge them, but I cant. It is very specific but yes ...

pavolNovak, 2023-04-26:
[image]
So First you have this situation  2 parts ( 1 of them consists of two components you want to print together cylinder+ball). Raft on the bottom is overlapped which will result in bad print. to accomodate this issue you can merge 2 of them together 
[image]
as you see the raft is fixed. But now you need to change something in the design but you cant. Suggested was to use split function which will result in ball droping to the build plate which is not what was intended. Therefore Unmerge function is needed.
[image]

pavolNovak, 2023-04-26:
[merge.zip](https://github.com/prusa3d/PrusaSlicer/files/11330558/merge.zip)

neophyl, 2023-04-26:
What are you actually trying to do ?  With your project I could update any of the parts individually,
[image]

For example I changed one of the cylinders with a random fan shroud I had on disk.  Just used the right click and replace with stl option.  The Reload from disk option is also available for individual parts too.  So you can update elements of the merged Object without effecting others.

The 'bad' part of it is that loading a different stl can position it strangely compared to where the original Part was but that's dependent on the 2 different stl files and is a separate  known issue with PS.  The coordinate system is in need of a general overhaul.

pavolNovak, 2023-04-26:
The coordinate system is then messed up yes ,and that was the main reason also to use this function somehow. What I tried to do Was printing a lot of small components that needed this raft hack and when I opened this project I needed to change the spacing between them ( arrange function) , but since it was already merged it doesn't work. There was also the problem if I would use (Split to object) it would mess up the parts, therefore I wanted to unmerge , rearrange and merge back again.

closed 2026-09-29 by prusaslicer-bot[bot]