prusa3d/PrusaSlicer #9006: Pressure Equalizer scales improperly

n8bot, 2022-10-06:
### Description of the bug

Pressure equalizer says it takes mm^3/s^2. However, when the feedrate is increased, the pressure equalizer dramatically reduces the allowed volumetric acceleration.

### Project file & How to reproduce

Load up any print profile, and set slopes to 1 mm^3/s^2. Slice, notice the print time, and the speed visualization. Then, increase the infill speed to something like 180 mm/s. Notice that the print time is actually LONGER.

The infill will now take longer to hit the same feedrate as before.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

win 10

### Printer model

any prusa profile

murk-sy, 2022-10-06:
You should share your project file, because I can't reproduce this with default Mini 0.20 SPEED profile. It seems to behave as it should.

n8bot, 2022-10-06:
1. Load mini 0.2 quality profile. 
1. Add a cube, scale it to 80x80x80 mm.
1. Configure positive and negative slope of `1 mm^3/s^2`
1. Slice. See this:

[image]
[image]
**Note that the requested feedrate of 80 mm/s is achieved around this point where the nozzle is shown.** (Ignore the fact that the feedrate fluctuates -- this is due to single precision float usage instead of double precision. Easy fix.)

5. Now, change the infill feedrate to 180 mm/s.
6. Slice. See this:

[image]
[image]
**Note that the requested feedrate of 180 mm/s is never achieved. Not even 80 mm/s is achieved by the time the infill starts to "decelerate" due to the pressure equalizer.** (Ignore the fact that the deceleration phase is abrupt -- this is due to the lookback value of 128. Expected.)

murk-sy, 2022-10-06:
Okay, yeah, important discovery: this happens on Prusament PLA, but *not* on Prusament PETG @MINI profile due to max volumetric speed. I forgot to even check actual speed reached, was only looking at volumetric speed view.

closed 2026-09-29 by prusaslicer-bot[bot]