prusa3d/PrusaSlicer #9008: Prusaslicer 2.5 mmu filament selection not respected

ikrEErki, 2022-10-06:
### Description of the bug

I split my own mmu model. After that, when changing the filament I want to use, it is not respected. Please look at the picture below. Color should be red, it is white. White is first tool, red is tool number 4. Also the gcode picks tool 0.

`;go outside print area
G1 Y-3 F1000
G1 Z0.4 F1000
; select extruder
T0
; initial load
G1 X55 E29 F1073
G1 X5 E29 F1800
G1 X55 E8 F2000
G1 Z0.3 F1000
G92 E0
G1 X240 E25 F2200
G1 Y-2 F1000
G1 X55 E25 F1400
G1 Z0.2 F1000
G1 X5 E4 F1000`

[image]

### Project file & How to reproduce

[Frys islucka v6_Frys islucka v6_Body1.zip](https://github.com/prusa3d/PrusaSlicer/files/9727408/Frys.islucka.v6_Frys.islucka.v6_Body1.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 10

### Printer model

Mark 3s with mmu2s

BaconFase, 2022-10-10:
I may help to describe every step you took (within PS) to get to this situation. Things seem to work correctly after deleting one of the parts (and then re-importing).

ikrEErki, 2022-10-11:
Well, I didn't do anything except what I wrote.

1. Import the model
2. Split model
3. Problem is there

closed 2026-09-29 by prusaslicer-bot[bot]