prusa3d/PrusaSlicer #9009: Add smooth transition between external perimeter and small perimeter speeds

murk-sy, 2022-10-07:
**Is your feature request related to a problem? Please describe.**
This is a fairly minor problem, but:
On thin and tall models with either width changing spikes or, as shown, curvy shapes, the transition between full External Perimeter Speed (ExPS) and Small Perimeter Speed (SmPS) is visible. It makes the part that is otherwise flawless look glued together. Glued together *extremely well*, but still.

[image]
[image]
[image]
[image]

**Describe the solution you'd like**
Ideally, SmPS should not be used *somehow* in cases like shown, or there should be a transition between speeds.

**Describe how it would work**
Assuming ExPS being 40 and SmPS being 25 at radius 6.5mm (as noted in description), the speed should smoothly interpolate between 40 at radius 10 mm, 32.5 at radius 8.25 mm and 25 at radius 6.5mm (though making it lower - or settable - might be a good idea).
Note that Max Volumetric slope of 2 is already used.

**Describe alternatives you've considered**
I could just manually change the SmPS setting to same as ExPS, but I don't know how this would affect really small holes - this model in particular doesn't have them.

**Additional context**
An example model for testing:
[image]

[Small perimeter speed test 2022-10-07.zip](https://github.com/prusa3d/PrusaSlicer/files/9731025/Small.perimeter.speed.test.2022-10-07.zip)

rtyr, 2022-10-07:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/2785.

aknezevi-work, 2025-05-09:
I honestly don't know why small perimeter speed overrules external perimeter speed. Small perimeter speed should only be used for inner perimeters, that is where it is actually useful.

murk-sy, 2026-09-24:
Reproduced in 3.0.0-alpha12

[2026-09-24 issue 9009 ps3.zip](https://github.com/user-attachments/files/32618363/2026-09-24.issue.9009.ps3.zip)