prusa3d/PrusaSlicer #9010: Modifier speed inconcistencies

rongith, 2022-10-07:
### Description of the bug

Speed change in a modifier does sometimes trigger a rebuild of the paths but not all kind of speed.
Modifier cut the paths in some cases, it is no more continuous when crossing the shell of the modifier.

### Project file & How to reproduce

If I change the external perimeter speed. The paths are not change. And it seems the speed of the tool doesn't really change.
If I change the Gap fill speed. The paths are changed.
And what I don't really like is that the fact that the external perimeter in some cases is no more continuous when crossing the modifier limit. The printed part is then weaker.
[image]
[TestLattice-r00c - 3D.zip](https://github.com/prusa3d/PrusaSlicer/files/9734020/TestLattice-r00c.-.3D.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Win11

### Printer model

PMK3S+

DragonFinlay, 2022-10-12:
Regarding the speed:
1. You are being bottlenecked by volumetric flow. Printing at 200 mm/s would far exceed the limit of 25 mm³/s that's set in the filament settings. These limits are important, they give you control and prevent your hotend from clogging up because it can't handle the filament flow.
2. The modifier mesh seems to have an issue as the print speed is raised properly if I apply the change globally instead of just for the modifier. Please double-check the modifier model and let me know the results.

[image]

Regarding the perimeters: Currently certain modifier changes can't be merged perfectly into the model yet. We should be able to improve this in the future.

DragonFinlay, 2022-10-12:
I see, I didn't realize that the point about speed was merely meant as a comparison.
In that case, it's really just the gap fill speed that is programmed to split the model to prevent issues (for now, as explained in the second point of my previous comment).

I agree though that changing the gap fill speed should in no way influence a print that doesn't even have gap fill (because Arachne is used). I'll forward this to the team, perhaps it's an easy fix.

rongith, 2026-09-07:
There is no more inconsistencies when changing the gap speed.
Though the desired result is not what I want, I think it is working as intended. Then I close the issue.

closed 2026-09-07 by rongith