prusa3d/PrusaSlicer #9013: Octoprint Stop Button

44knight44, 2022-10-08:
I like the idea of uploading and printing the part on my printer through Octoprint. But if it fails I would like to have a stop button so I don't have to open up octoprint and cancel. Just wondering if it is possible to do this. Thanks. 
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

dtryba, 2022-10-08:
The MK3S has that stop button, it is "hidden" under tune IIRC. With the Mini it is more obvious to just disconnect the serial input from the menu.

DragonFinlay, 2022-10-08:
There should always still be a stop button, both on the MK3S+ and the MINI+.
Anyway, this is the wrong repository to post this in. In case you're still looking to report this, please create an issue in [prusa3d/Prusa-Firmware](https://github.com/prusa3d/Prusa-Firmware) or [prusa3d/Prusa-Firmware-Buddy](https://github.com/prusa3d/Prusa-Firmware-Buddy) respectively. Thanks!

closed 2022-10-08 by DragonFinlay