prusa3d/PrusaSlicer #9017: Auto-center does not turn off

jonathanmwagstaff, 2022-10-08:
### Description of the bug

After turning off the Auto-center[ing], the imported stl is still centered on the build plate and not located relative to its origin

### Project file & How to reproduce

Configuration -> Preferences -> General -> Auto center parts
make sure that the checkbox is unchecked

import an stl with the origin located near the object

the object is still centered at the center of the build plate regardless of the location of the part in the STL file.

[image]

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 11 Home 21H2 64 bit

### Printer model

Ender 3

jonathanmwagstaff, 2022-10-08:
I want to slice a shape and add the gcode to a custom gcode shape. In order to do this I need to know where the origin is relative to my sliced shape

rtyr, 2022-10-10:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/1763. I am closing this one.

closed 2022-10-10 by rtyr