prusa3d/PrusaSlicer #9018: Adding option to duplicate modifier, negative volume, and support blocker and enforcer

pedjas, 2022-10-08:
**Is your feature request related to a problem? Please describe.**
Often I have models that are symmetrical and I have to add several modufiers/support enforcers that are the same size and shape except position. There is no option to duplicate existing shape so I have to create each one from scratch.

**Describe the solution you'd like**
Option to duplicate modifier, negative volume, and support blocker and enforcer.

**Describe how it would work**
Right click on shape and select option Duplicate. Duplicate of the shape is created.

**Describe alternatives you've considered**
I found no alternatives other than creating the same shape manually, over and over again.

neophyl, 2022-10-08:
Select just the modifier and then use ctrl+C to copy and ctrl+V to paste

rtyr, 2022-10-09:
Answered. Closing.

closed 2022-10-09 by rtyr

pedjas, 2022-10-09:
> Select just the modifier and then use ctrl+C to copy and ctrl+V to paste

That works? But why there is no menu option?