prusa3d/PrusaSlicer #9019: PS 2.5.0 Arachne Engine detail

kgolger, 2022-10-08:
### Description of the bug

I did a little experiment with PS 2.5.0 regarding the perimeter settings. I printed a small part with 3mm lettering (part size max. approx. 30mm) with a 0.25 nozzle. I noticed in advance that in small, detailed sections perimeters are calculated very cryptically (see example on both following images):
[image]
(2.5.0_Arachne_Perimeter)

[image]
(2.5.0_Classic_Perimeter)

### Project file & How to reproduce

[fanshroud_ring_v10_ASA.zip](https://github.com/prusa3d/PrusaSlicer/files/9740352/fanshroud_ring_v10_ASA.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

W10 Pro 64-bit

### Printer model

Prusa Mini, FW 4.2.0

kubispe1, 2022-10-09:
I think this could be similar to #8730
[image]

kgolger, 2022-10-09:
Thanks for the very interesting and helpful link!
 
wbr,
Karl

 

 <https://user-images.githubusercontent.com/61363580/194768186-50986a4c-c38f-4caa-859f-09228c7f68d8.png>

kubispe1, 2022-10-10:
I discussed with developer and it is really other example of attached issue. Thank you @kgolger. I am closing this issue as duplicate.

closed 2022-10-10 by kubispe1